system-linux/FazJammer #4: Jammer Error

exoryzen, 2025-04-21:
i have followed all the steps, however when i turn on my esp, the notification appears (Jammer Error), i also use the flash button as a push button can this also cause the problem?

if you have a solution please contact me
exoryzen@proton.me

system-linux, 2025-04-25:
Did you slove the issue?

exoryzen, 2025-04-26:
> Did you slove the issue?

totally no, Maybe I need to see your wiring if you don't mind sending it by Email or Telegram, i really need your help :(

system-linux, 2025-04-28:
I don't have the jammer with me right now. But the probable cause is that the SPI pin conflicts data when trying to manage two modules. Normally i solved this in code. Are you using ESP8266 or another microcontroller?

exoryzen, 2025-04-28:
> I don't have the jammer with me right now. But the probable cause is that the SPI pin conflicts data when trying to manage two modules. Normally i solved this in code. Are you using ESP8266 or another microcontroller? If you are using another microcontroller this shoul be related to it.

saya menggunakan ESP8266 Oled (HW-364B)

system-linux, 2025-04-30:
Try to use other button except reset or boot button. Maybe it can reset yourself when you press it.

janwego, 2025-04-30:
May I ask - is it intended that D5 (GPIO14) is used for SCK and SDA? As well as D6 (GPIO12) which is used for MISO and SCL?

Jaidm99, 2025-05-06:
> I don't have the jammer with me right now. But the probable cause is that the SPI pin conflicts data when trying to manage two modules. Normally i solved this in code. Are you using ESP8266 or another microcontroller? If you are using another microcontroller this shoul be related to it.

How to solve the isse? Do i need to change the code for nrf?

Miki104, 2025-05-11:
Have you found a solution? I was thinking of removing the string of code that shows the text on the display (line 98), maybe that could solve the problem, going directly to the next string. Can someone try it so maybe we can see if it works? Because im not sure if that works, it may also get stuck on the next line of code.

janwego, 2025-05-11:
To be honest I re-did the code and used the ports for SDA / SDL / MOSI / MISO which are recommended for my board (there are different esp32, esp8266 board with different pin-schemes). 
I simply updated the code (You have to find out if you have to address the pins as D4 or in the GPIO-style). Afterwards it worked right away.

Miki104, 2025-05-11:
What board did you use? The one that used the creator or another board?

Miki104, 2025-05-11:
What board did you use? The one that used the creator or another board? So i need just to test the pins if they are GPIO-style or D4 and it works?

janwego, 2025-05-11:
I used the ESP-12F Board. You have to find out which board you use and then search for google images with "YOUR_BOARD_NAME pin out". That will give you a drawing about what pin should be used for what.
It's surely not trivial and no beginner stuff but I can't guide you through that. 
Then you need to modify the Script with the correct ports, compile, load to your device and cable accordingly.
BR

Miki104, 2025-05-11:
Okay, thank you. Btw can you help me with my other issue? I dont have that much experience with ESP32, i'm a beginner so i sont know very much about this

Jaidm99, 2025-05-11:
I have fix the stuck issue. I change the pinout of the oled. It get conflicts when 2 module uses the same pin for the oled and nrf. But now i've got jammer error on the screen. It is really posible to work this jammer on esp8266? Even Ai says it imposible to for esp8266 to make a project for bluetooth jamming. Sorry, just a newbie here.

janwego, 2025-05-11:
Yes, sorry but that really is a newbie question :D
YES it is not possible to make a bluetooth jammer on an esp8266 as it does NOT have Bluetooth. But this project uses an external Antenna which handles the bluetooth frequencies and the esp8266 just controls this antenna.

BR

Jaidm99, 2025-05-11:
> Yes, sorry but that really is a newbie question :D YES it is not possible to make a bluetooth jammer on an esp8266 as it does NOT have Bluetooth. But this project uses an external Antenna which handles the bluetooth frequencies and the esp8266 just controls this antenna.
> 
> BR
Did you make it work on a esp8266?  Maybe i should i'll just use a esp32 instead. Having hard times making this to work on my esp826 😅

cviper13, 2025-05-12:
If the issue is solved, I'll ask the repo owner to close it. If there's any more issues **regarding this issue** ask me here. If you have more issues that isn't about this issue open a new issue. Thanks.

cviper13, 2025-05-26:
**Closing due to inactivity.**
No response for two weeks after follow-up. If the issue persists, feel free to comment here to reopen it. For unrelated issues, please open a new one. Thanks!