system-linux/FazJammer #7: I dont understand how to wire the OLED to ESP8266

Miki104, 2025-05-11:
In the wiring description it says that i have to use D5 and D6 but they're already in use from the antenna. Should i buy a module like this one?: https://www.amazon.it/NodeMCU-ESP8266-sviluppo-display-Micropython/dp/B0D9FP7RHK/ref=mp_s_a_1_5?crid=GR93VJ78RXGU&dib=eyJ2IjoiMSJ9.WAw3fn3viiwXrTUhcfqy6bGNof3ad5HZYWaA0BpCvE_f1Ysr14hqwdfl_GuImHSUSlht6u4J77PbpUyqqr3c07n7oRGW17Qb1DK_bP6ABN_evl1IE9XAlQzRvWio_6Gy1RQtlITbh9w8Di9Ml6G0CB-dzQoptrCrxGvJ5j5eHLiTzuVr3U3FOA1V4zFnElyfX03_7jRkpwyVB4WHztocLg.dvtvCyOXz-lngKkFakgq-MKXYci7_7U05hF0JsHY9nY&dib_tag=se&keywords=esp8266+nodemcu+con+oled+saldato&qid=1746976040&sprefix=esp8266+nodemcu+con+oled+saldato%2Cindustrial%2C120&sr=8-5

janwego, 2025-05-11:
That's exactly what I said was wrong. The code uses a pin twice, this will most likely not work. Find which pin should be used for which function, change it in the script and cable it

cviper13, 2025-05-12:
ESP8266 has only one hardware SPI pin. The code uses the same pin to drive the both devices (screen and 2.4GHz module). When 2.4GHz module needs to be used the communications with OLED screen is cut and communications with 2.4GHz module is done, vice versa for the OLED screen. Because that both devices has a chipset onboard they don't cut operations.

TL;DR: Both devices use the same SPI pin because theres only one on ESP8266.

closed 2025-05-17 by system-linux