system-linux/FazJammer #11: BLE Jammer couldn't be started!

McVillano, 2025-05-28:
I keep getting this message via serial console, i've triple checked the cable conections and are all according to you diagram. Also checked with a tester if the cables are okey and i get conections between the NRF24L01 Pins and the nodemcu Pins.

I've added a "radio.printdetails()" sentence for get more info. This is my serial console output;

`SPI Speedz	= 10 Mhz
STATUS		= 0xff RX_DR=1 TX_DS=1 TX_DF=1 RX_PIPE=7 TX_FULL=1
RX_ADDR_P0-1	= 0xffffffffff 0xffffffffff
RX_ADDR_P2-5	= 0xff 0xff 0xff 0xff
TX_ADDR		= 0xffffffffff
RX_PW_P0-6	= 0xff 0xff 0xff 0xff 0xff 0xff
EN_AA		= 0xff
EN_RXADDR	= 0xff
RF_CH		= 0xff
RF_SETUP	= 0xff
CONFIG		= 0xff
DYNPD/FEATURE	= 0xff 0xff
Data Rate	= 1 MBPS
Model		= nRF24L01
CRC Length	= 16 bits
PA Power	= PA_MAX
ARC		= 15
BLE Jammer couldn't be started!
`

Have you got a clue on what's happening?

McVillano, 2025-05-28:
I’ve also tried to comment the screen set up and first message, to see if it was a SPI problem because it was ocupied but still getting serial Ble error mensaje.

Also tried to prompt a messaje of ble error on the screen after serial mesage error but it seems that when the radio.begin() fails, blocks the SPI

cviper13, 2025-05-29:
Looking at your serial output, I can see the issue clearly. All the register values are showing as 0xff, which is a classic symptom of SPI communication failure with the NRF24L01+ module. Here's what's happening and how to fix it:

The Problem

When you see all register values as 0xff (especially STATUS = 0xff), it means the ESP8266 isn't actually communicating with the NRF24L01+ module. This is typically caused by:

Power supply issues (most common)

Wiring problems

Defective NRF24L01+ module

Solutions to Try

1. Power Supply Fix (Most Important)

The NRF24L01+ is very sensitive to power supply quality

```
// Add this to your setup() before radio.begin() delay(100); // Give power time to stabilize 
```

Hardware fixes:

Add a 10µF capacitor between VCC and GND on the NRF24L01+ module

Add a 100nF ceramic capacitor as well for high-frequency noise

Use a separate 3.3V regulator if possible (NodeMCU's 3.3V can be weak under load)

Keep wires as short as possible (especially power wires)

The key issue is almost certainly power supply related. The NRF24L01+ draws significant current spikes during transmission, and the NodeMCU's 3.3V regulator often can't provide clean, stable power.

Try the power supply fixes first - that solves this problem in 90% of cases!

McVillano, 2025-05-30:
> Looking at your serial output, I can see the issue clearly. All the register values are showing as 0xff, which is a classic symptom of SPI communication failure with the NRF24L01+ module. Here's what's happening and how to fix it:
> 
> **The Problem**
> 
> When you see all register values as 0xff (especially STATUS = 0xff), it means the ESP8266 isn't actually communicating with the NRF24L01+ module. This is typically caused by:
> 
> **Power supply issues (most common)**
> 
> * Wiring problems
> * Defective NRF24L01+ module
> 
> **Solutions to Try:**
> 
> 1. Firmware Fix:
> 
> The NRF24L01+ is very sensitive to power supply quality
> 
> ```
> // Add this to your setup() before radio.begin() 
> delay(100); // Give power time to stabilize 
> ```
> 
> 2- Hardware fixes:
> 
> Add a 10µF capacitor between VCC and GND on the NRF24L01+ module
> 
> Add a 100nF ceramic capacitor as well for high-frequency noise
> 
> Use a separate 3.3V regulator if possible (NodeMCU's 3.3V can be weak under load)
> 
> Keep wires as short as possible (especially power wires)
> 
> The key issue is almost certainly power supply related. The NRF24L01+ draws significant current spikes during transmission, and the NodeMCU's 3.3V regulator often can't provide clean, stable power.
> 
> Try the power supply fixes first - that solves this problem in 90% of cases!

I've tried several things:

The daly is not necesary as i'm making test with only the piece of code that operates with radio, just to test the NFR24L01
Today I've received 2 more units for another project and i can ensure it's not a daulty module, with the other ones is the same problem.

I've reduced the wiring to a minimal 10 cm dupont cables and still the same.

I've used the 10µF and the 100µF capacitors (separated and even both ones) to reduce noise and still the same

I've tested the NFR24L01 with a test code on a arduino nano V3 and radio.begin() outputs a true, but i've not been able to do radio.printdetails() i dont know if that order works in the arduino nano

I've measured power both in the Nodemcu and the Nano and the nodemcu is providing 3.28 V both in the board pins and the NRF24L01 Pins , and the Nano was producing 3.5V in the board pins and 3.4V in the NRF24L01 pins

So i suppose that is the problem, i tried to adapt the code from the repo to arduino nano to use that board but the screen doesn´t do anything and the serial monitor only prompts "OL" these two leters

I don't want to buy a external power balanced board so i supose this is al for now, if i think of new tests or solve the problem somehow i will post it here

Thanks for everything

system-linux, 2025-06-09:
Can you change D2 and D4 pins.

closed 2025-06-24 by cviper13