system-linux/FazJammer #16: No transmission

chrisground, 2025-06-21:
Hello, thanks for your project, I think I did everything right, get no errors, but look like the RF24 is not transmitting anything, doesnt affect wifi nor bluetooth, and stays cold.

cviper13, 2025-06-21:
Hello, could you try using the RF24 with a generic project, for example connect it to your phone via bluetooth, to see if it works on your current setup. If doesn't then its a circuit issue. If it does its an issue on thr code and we will debug with you.

ShadowNetworkSNC, 2025-08-17:
I am also having this issue the jammer comes on lets me select modes but does not transmit anything. i have also tried another rf24. 

> Hello, could you try using the RF24 with a generic project, for example connect it to your phone via bluetooth, to see if it works on your current setup. If doesn't then its a circuit issue. If it does its an issue on thr code and we will debug with you.