system-linux/FazJammer #20: NodeMCU ESP8266 OLED doesn't seem to react to Adafruit libraries

Pyrobotx2300, 2025-06-30:
Using what appears to be the same board that the guide was using, after successfully compiling and uploading to the board, no activity is observed on the OLED screen.

**Product Used:**
https://www.amazon.com/Development-Wireless-Micropython-Programming-Soldered/dp/B0BVM3H46B/ref=asc_df_B0BVM3H46B?mcid=298c44b5b2ae3fe4a397fffbd2ee5781&hvocijid=17537349762150593426-B0BVM3H46B-&hvexpln=73&tag=hyprod-20&linkCode=df0&hvadid=721245378154&hvpos=&hvnetw=g&hvrand=17537349762150593426&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9007814&hvtargid=pla-2281435177858&psc=1

**Software and Libraries available:**
Arduino IDE 2.3.6

Adafruit BusIO - 1.17.1
Adafruit GFX Library - 1.12.1
Adafruit SSD1306 - 2.5.15
RF24 - 1.5.0
U8g2 - 2.35.30 (using this library, I can populate the OLED)
ezButton - 1.0.6

**Installed Boards:**
Arduino ESP32 Boards - 2.0.18-arduino.5
esp32 - 3.2.0
esp8266 - 3.1.2

In addition, following the board listings' instructions, I have installed the CH340 driver on my computer thus providing me access to upload code to the device.

The only other thing of note that I could find was on the Adafruit SSD1306 github, there is a note saying that for the ESP8266 boards: _Change OLED_RESET to different pin if using default I2C pins D4/D5._
Now, since the OLED came preinstalled I haven't looked into if this is something that is possibly a bottleneck but my assumption is that this is not something needed to look into since the guide didn't mention it.

Attached is an image of how I have the device set up showing no display but having power. If requested I can also attach my wiring configuration, however, i'm pretty confident that it is accurate to the instructions in the guide.
[image]

Pyrobotx2300, 2025-07-03:
> Hey, if you look at the pinout for this board you've linked, you'll see the GPIO 12 and 14 are flipped to the instructions on this repo.
> 
> To fix, flip this line:
> 
> `Wire.begin(14, 12); //SDA, SCL` to `Wire.begin(12, 14); //SDA, SCL`

This absolutely did the trick. This was my first time using a micro-controller so I genuinely didn't know what I was looking at lol. Thanks so much!

system-linux, 2025-07-06:
> Hey, if you look at the pinout for this board you've linked, you'll see the GPIO 12 and 14 are flipped to the instructions on this repo.
> 
> To fix, flip this line:
> 
> ` Wire.begin(14, 12);  //SDA, SCL `
> to  
> `Wire.begin(12, 14);  //SDA, SCL`

Hey, I didn't notice that. Thanks, i will commit :D